devbisme/WireIt #1: New features / enhancements suggestions #1

gh-jmg, 2018-06-21:
New features / enhancements suggestions (from KiCad forum)

1) Can you add ability to edit a netname ? eg if you make a typo, and need to correct. I guess this needs to warn, if the new name results in a net-merge.

2) Currently it brick-walls on merge of nets, can that enhance to give the choice to use either name ?
(ie optionally allow net merge?) Some CAD flows give the 3rd choice of enter-net name when merging nets.

3) Currently, if it finds a net name, on any PAD, it applies that.  A nice feature.
If you select N-Pads(no names) and >= 1 NET segment, can it ‘seed’ the NET name, from that selected segment ?
ie This extends that Name-seeding IQ very slightly, to checking if anything selected has a net name.

Minor bug/quirk:  Set Net name window, does not accept <enter> you must click [OK] - can it be fixed to allow <enter> ?

xesscorp, 2018-06-21:
I fixed the net name window to focus on the OK button after a net name is entered.

xesscorp, 2018-06-25:
Net merging and renaming are now supported.

gh-jmg, 2018-06-26:
That looks great, 1. and 2. are nicely enhanced. Thanks.

3. is still not there ?

ie if I select a named track segment, and then a PAD, it does not seed the net name from the selected track ?
I can use the new drop-down list-of-nets, but on a larger board, that's more searching/matching.