devbisme/WireIt #4: Icons disappeared, tool won't run

robert128, 2018-06-21:
Hello,

I love this plugin!  Unfortunately, after swapping a few pins and then doing some routing, the icons have disappeared, and when I try runing WireIt again, nothing happens.  I'm running a nightly build (June 19) on Windows 7.  Is there some way to debug it?  Perhaps from the scripting console?  Even after quitting Pcbnew and restarting, I can't get WireIt to run, I have to quite Kicad completely.

xesscorp, 2018-06-21:
I'm using Windows 7 and a June 21st nightly. I'm not seeing anything out of the ordinary. I'll probably need to get your .kicad_pcb file to determine if the problem is in your environment or caused by some interaction between your design and WireIt.

Have you tried WireIt with any other PCBs and seen the same behavior?

robert128, 2018-06-21:
I've been running on the same design with a different machine (Windows 10) and June 21 Kicad for a few hours now without any troubles except swapping pins is still painfully slow.  I've also been able to save the changes properly too.  When I manually make changes, they don't take any time, so I have hope that swapping pins with WireIt shouldn't actually make my computer go away for a long time.

xesscorp, 2018-06-22:
To check the speed of the pin swap, try the following:

1. Open the Python scripting console.
2. Type the command `from pcbnew import *`.
3. Select two pads in your board.
4. Now type the following sequence of commands:
```py
    brd = GetBoard()
    pads = [p for p in brd.GetPads() if p.IsSelected()]

    pad0_net = pads[0].GetNet()
    pads[0].SetNet(pads[1].GetNet())
    pads[1].SetNet(pad0_net)

    brd.BuildListOfNets()
    brd.GetConnectivity().RecalculateRatsnest()
    Refresh()
```
If the pad swapping is taking a long time, then it should be apparent when running the above sequence of commands where that time is being spent.

robert128, 2018-06-22:
When I type in the commands, I only noticed a delay after the Refresh(), but typing is slow and I'm not playing with the layout to notice delays.  I can't explain it, but I've tried twice to paste all but the Refresh() command into the scripting console, and both times, there was no swap.  But I did see a delay after Refresh().  I'd put my money on the Refresh().

xesscorp, 2018-06-22:
You should type in the commands one-by-one and observe any delays that occur. If there is no delay in displaying the prompt after typing in a command and hitting return, then that command couldn't be responsible for the 30s delay.

Also, a pin swap wouldn't be apparent until the Refresh() operation is applied.

Refresh() is exposed through the KiCad Python API. It's just a call into the KiCad screen refresh code. If you're not seeing a delay when doing normal PCBNEW stuff (like zoom in/out), then I don't know why you would see a 30s delay when calling Refresh() within WireIt.

The design I test with has 150 nets and 600-700 pads. How large is your design? Have you got anything on your board that's out of the normal, like strange copper zones or large silkscreen graphics?

robert128, 2018-06-22:
The delay I saw was only apparent after Refresh().

I'm on a different machine that has much less delay, but using your tip that graphics might be involved, I switched from Accelerated graphics to Fallback, and I no longer see the delay.  I don't see any delays at any other time, so there is something about doing Refresh() in Accelerated mode.

gh-jmg, 2018-06-22:
> I love this plugin! Unfortunately, after swapping a few pins and then doing some routing, the icons have disappeared, and when I try runing WireIt again, nothing happens

In case it helps, when first installing WireIt, I had missing icon symbols, but the mouse hover worked & operations worked. That was caused by me miss-reading the instructions, and I had placed the icons in the same dir as the .py file. 
When I preserved the subdirectory relative structure of the ZIP, icons appeared, and have never gone away since. (Win10, recent V5 build)

xesscorp, 2018-06-23:
So with accelerated graphics disabled, is the pad swapping delay gone when you are using WireIt normally (i.e., not typing commands into the Python shell)?

closed 2018-06-25 by xesscorp

robert128, 2018-06-26:
It's working normally on my newer machine without delay, but I haven't had a chance to try it on the older one.