devbisme/WireIt #5: Crash with pcbnew or footprint editor with Kicad 5.0 on Ubuntu 18.04.1

ngocminhdao88, 2018-08-01:
**Describe the bug**
PCB layout editor and Footprint library editor (Kicad 5.0) crashes with WireIt plugin on Ubuntu 18.04.1

**To Reproduce**
Steps to reproduce the behavior:
1. Open Kicad 5.0
2. Click on PCB layout editor or Footprint library editor
3. It crashes

**Computing Environment**
Application: kicad
Version: 5.0.0-fee4fd1~65~ubuntu18.04.1, release build
Libraries:
    wxWidgets 3.0.4
    libcurl/7.58.0 OpenSSL/1.1.0g zlib/1.2.11 libidn2/2.0.4 libpsl/0.19.1 (+libidn2/2.0.4) nghttp2/1.30.0 librtmp/2.3
Platform: Linux 4.15.0-29-generic x86_64, 64 bit, Little endian, wxGTK
Build Info:
    wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8) GTK+ 2.24
    Boost: 1.65.1
    OpenCASCADE Community Edition: 6.9.1
    Curl: 7.58.0
    Compiler: GCC 7.3.0 with C++ ABI 1011

Build settings:
    USE_WX_GRAPHICS_CONTEXT=OFF
    USE_WX_OVERLAY=OFF
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=ON
    KICAD_SCRIPTING_WXPYTHON=OFF
    KICAD_SCRIPTING_ACTION_MENU=ON
    BUILD_GITHUB_PLUGIN=ON
    KICAD_USE_OCE=ON
    KICAD_USE_OCC=OFF
    KICAD_SPICE=ON

**Additional context**
PCB layout editor or Footprint library editor does not crash without WireIt plugin.

xesscorp, 2018-08-02:
The Python plugins in KiCad 5.0 under Linux are broken because it depends on wxGTK2 but wxGTK3 is what is installed on Linux. You'll have to wait for a new maintenance release of KiCad before any Python plugins will work on Linux.

ngocminhdao88, 2018-08-02:
Oh i see. Thank you for the answer.

GaryMatthews, 2018-10-27:
has this problem been resolved yet. packaged version of kicad crashes as well as compiled version. need python console back up and running again.

xesscorp, 2018-10-29:
I just tried WireIt with KiCad 5.0.1-3 release build for Windows. Seems to work fine. No idea what it does on the various linuxes or OSX.

closed 2019-02-17 by xesscorp

rgbbytes, 2020-09-05:
Just because it works fine for you on Windows doesn't mean it's not a problem that it crashes in OSX!

xesscorp, 2020-09-05:
I never said it wasn't a problem for them. But, my motivation for building WireIt was for my own use. I released it in case it might be useful for others. Nowhere in there did I make a commitment to get it working for *everyone on every platform*, particularly OSX. It's open source, so if you really need it you can try to get it running.

You might want to take a look at some of the [closed issues](https://github.com/xesscorp/WireIt/issues?q=is%3Aissue+is%3Aclosed) to see that I have helped users fix some problems with WireIt. But I'm not obligated to do so.