devbisme/WireIt #9: Icons appear in wrong location

ngocminhdao88, 2019-05-04:
**Describe the bug**
Icons of WireIt tool appear in KiCad project explorer, not in PCBnew

**To Reproduce**
1. Go to PCBnew
2. Click on Tool -> External Plugins -> WireIt
3. Screen flash, WireIt's icons don't appear in PCBnew but in KiCad project explorer

**Expected behavior**
Icons should be appeared in PCBnew software.

**Screenshots**
[image]

**Computing Environment**
Application: kicad
Version: 5.1.2-f72e74a~84~ubuntu18.04.1, release build
Libraries:
    wxWidgets 3.0.4
    libcurl/7.58.0 OpenSSL/1.1.0g zlib/1.2.11 libidn2/2.0.4 libpsl/0.19.1 (+libidn2/2.0.4) nghttp2/1.30.0 librtmp/2.3
Platform: Linux 4.15.0-48-generic x86_64, 64 bit, Little endian, wxGTK
Build Info:
    wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.22
    Boost: 1.65.1
    OpenCASCADE Community Edition: 6.9.1
    Curl: 7.58.0
    Compiler: GCC 7.3.0 with C++ ABI 1011

Build settings:
    USE_WX_GRAPHICS_CONTEXT=OFF
    USE_WX_OVERLAY=ON
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=ON
    KICAD_SCRIPTING_PYTHON3=ON
    KICAD_SCRIPTING_WXPYTHON=ON
    KICAD_SCRIPTING_WXPYTHON_PHOENIX=ON
    KICAD_SCRIPTING_ACTION_MENU=ON
    BUILD_GITHUB_PLUGIN=ON
    KICAD_USE_OCE=ON
    KICAD_USE_OCC=OFF
    KICAD_SPICE=ON

xesscorp, 2019-05-06:
I checked this on KiCad 5.1.2 for Windows. It works correctly. Perhaps this is somehow related to the problems KiCad was having supporting wxPython under linux?

ngocminhdao88, 2019-05-07:
I just check this plugin with KiCad 5.1.2 on virtual Windows machine. It works like intended. So this is a bug of KiCad on linux.

Is there a workaround for linux user beside of waiting for KiCad team fix the wxPython problem?

holmesrichards, 2020-07-01:
I see this behavior in KiCad 5.1.5 under Linux Mint.

andypfau, 2020-07-22:
Same issue on Fedora 32. I dug a bit into the code; the problem is in line 730 of WireIt.py:

> `top_toolbar = pcbwin.FindWindowById(ID_H_TOOLBAR)`

If I change `ID_H_TOOLBAR` to `ID_V_TOOLBAR`, `ID_AUX_TOOLBAR` or `ID_OPT_TOOLBAR`, it appears in the corresponding toolbar of the Pcbnew window. So I'd guess that the ID_H_TOOLBAR constant is incorrectly defined?

If found the constant in line 8212 of [pcbnew.py](https://kicad-downloads.s3.cern.ch/doxygen-python/pcbnew_8py_source.html), but since I'm not experience with KiCad development I have no idea where to look next.

xesscorp, 2020-07-23:
WireIt was using the wrong parent window when searching for the PCBNEW horizontal toolbar so it was placing the icons in the main KiCad window. I fixed that and it appears to be working now.

closed 2020-07-23 by xesscorp