devbisme/WireIt #11: Plugin does not register

blinkenlight, 2019-11-21:
I'm running Kicad version (5.1.4)-1 on Windows 7. I copied WireIt.py (and its icon folder) into my Kicad plugin folder (next to the other plugins I use and which work fine), under "C:\Users\Username\AppData\Roaming\kicad\scripting\WireIt" but Layout was not seeing it no matter how many times I clicked "refresh" under the "external plugins" menu, until I added next to it a file named "__init__.py" with the following content:

```
from __future__ import print_function
import pprint
import traceback
import sys

print("Starting plugin WireIt")
try:
    from .WireIt import *
    WireIt().register()
except Exception as e:
    traceback.print_exc(file=sys.stdout)
    pprint.pprint(e)
```

After that, everything worked fine. I didn't come up with the idea by the way - all my working plugins seem to come with a file like that, except the one other plugin that also fails to show up. So... it was a bit of a dead giveaway. Then again, some might not figure this out...

andypfau, 2020-07-22:
I tested your code under Fedora 32, where I had the same issue, and it works without modifications.

xesscorp, 2020-07-23:
I'm unsure how this would work with multiple plugins since they all go into the same `plugins` directory. Wouldn't their individual `__init__.py` files overwrite each other?

blinkenlight, 2020-07-23:
Not really, as [1] clearly mentions plugins with multiple files can go into their own sub-folder under _plugins_. I don't remember whether I tried at all to use WireIt _directly_ in the _plugins_ folder at the time of the original post when I couldn't get it to show up, but if one is to place it under its own folder, the __init__.py definitely needs to be there for anything to happen according to [1].

[1] - https://kicad-source-mirror.readthedocs.io/en/latest/Documentation/development/pcbnew-plugins/

xesscorp, 2020-07-23:
When I originally created WireIt, I didn't know there was the option of placing everything in its own subfolder. You're right: in a subfolder, it needs a `__init__.py` file. Here's the one I used:
```py
from . import WireIt
```
That seems to be all that's needed.

closed 2020-07-23 by xesscorp