devbisme/WireIt #12: pcbnew freezes after using WireIt tool  (Mac version of KiCad)

subatomicglue, 2020-04-24:
**Describe the bug**
pcbnew freezes after using WireIt tool

**To Reproduce**
Steps to reproduce the behavior:
1. Install:  
```
$ git co https://github.com/xesscorp/WireIt.git
$ cd WireIt
$ cp -r WireIt.py  WireIt_icons /Applications/KiCad/kicad.app/Contents/SharedSupport/scripting/plugins/
```
2. Tools / External Plugins / WireIt
3.  Select a few pads, wires, polygons for the ground net GND.   This also happens with at little as just 2 pads selected inside of a package (Module / Arduino_UNO_R3).
4.  Click the WireIt button
5.  Type in GND into the box, and hit ok.
6.  Try to do anything at all - but you can't - MacOSX emits an error tone every click.  
7.  Have to Force Quit the app

**Expected behavior**
I would expect the app to be functional after adding a net name to these elements.   There is no prompt or way to exit this frozen mode.

**Computing Environment** 
MacOS  Mojave 10.14.6 (18G95)

KiCad Pcbnew info:
```
Application: Pcbnew
Version: (5.1.5-0-10_14), release build
Libraries:
    wxWidgets 3.0.4
    libcurl/7.54.0 LibreSSL/2.6.5 zlib/1.2.11 nghttp2/1.24.1
Platform: Mac OS X (Darwin 18.7.0 x86_64), 64 bit, Little endian, wxMac
Build Info:
    wxWidgets: 3.0.4 (wchar_t,STL containers,compatible with 2.8)
    Boost: 1.69.0
    OpenCASCADE Community Edition: 6.9.1
    Curl: 7.54.0
    Compiler: Clang 9.0.0 with C++ ABI 1002

Build settings:
    USE_WX_GRAPHICS_CONTEXT=ON
    USE_WX_OVERLAY=ON
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=ON
    KICAD_SCRIPTING_PYTHON3=OFF
    KICAD_SCRIPTING_WXPYTHON=ON
    KICAD_SCRIPTING_WXPYTHON_PHOENIX=OFF
    KICAD_SCRIPTING_ACTION_MENU=ON
    BUILD_GITHUB_PLUGIN=ON
    KICAD_USE_OCE=ON
    KICAD_USE_OCC=OFF
    KICAD_SPICE=ON
```

xesscorp, 2020-04-26:
I've attached a WireIt.py file with some debugging statements. Try it and maybe we can see where PCBNEW is freezing.
[WireIt.py.gz](https://github.com/xesscorp/WireIt/files/4536416/WireIt.py.gz)

xesscorp, 2020-05-05:
I assume this is no longer of interest so the issue is closed.

subatomicglue, 2020-05-09:
If I cancel the dialog, using the new debug version.  It hangs in the same way....

subatomicglue, 2020-05-09:
should I be copying the script to `~/Library/Preferences/kicad/scripting/` instead?

subatomicglue, 2020-05-09:
Feels like kicad for MacOS has a bug when the script ends...  no matter OK or Cancel, once done we get the app stealing all mouse button input and emitting an error tone...

xesscorp, 2020-05-12:
OK, it looks like the script is getting lost inside the path for adding a new net and isn't even getting to the board update section of the callback. I've added some new debug dialogs to explore the program execution. Just repeat what you did last time and we'll see where the script stops responding.

[WireIt.py.gz](https://github.com/xesscorp/WireIt/files/4615594/WireIt.py.gz)

subatomicglue, 2020-05-17:
With the new plugin...

this time I'm running on a PCB that has some nets from a schematic...  highlighted two GND pads, and then clicked WireIt...   still the same result...

trace:
```
wire_it_callback init complete...
wire_it_callback case #2 
Rename Nets Attached to Pads   (type GND into the box, press OK)
```
Anyplace I click results in the error tone.

If I hit ESC, my pads deselect.  But that's the only result I can cause to happen.

xesscorp, 2020-05-17:
You're now running a different test case (case #2). I don't have any debugging built into that section of the code. You need to return to your first example (which displays case #1) and run that. The debugging statements I added will tell us more about where the code is getting lost.

subatomicglue, 2020-05-18:
Actually I ran both test cases.  See the 2nd half of my last post ;-)

xesscorp, 2020-05-18:
My apologies. The email notification didn't show the lower half of your message and I neglected to read it when I went to GitHub.

From the debug messages, it looks like WireIt.py is not even getting out of the modal dialog that's used to enter the net name. I modified that code, so maybe we can get further this time...
[WireIt.py.gz](https://github.com/xesscorp/WireIt/files/4645112/WireIt.py.gz)

subatomicglue, 2020-05-18:
Ok, latest (running on a PCB without any schematic)

```
wire_it_callback init complete...
wire_it_callback case #1
Attach Pads To Nets   (type GND into the box, press OK)
Net namer completed...
A new net name was entered...
Adding pads to net...
wire_it_callback building list of nets...
wire_it_callback recalc rats nest...
wire_it_callback refresh...
wire_it_callback DONE!
```

Strange enough - the script is working now.    That's odd!   I see a virtual wire between the two pads now, and I can interact with the design now (no error tones)...

Just verified that the original `WireIt.py` from github fails in the old way.   Diffing the two, there are only debug statements, so maybe there is a bug in kicad windowing where it gets unstuck if you exercise the dialogs more...  (total shot in the wind on that idea)

Diff
```
> diff WireIt-debug.py WireIt-github.py
413c413
<         self.Close()
---
>         self.Destroy()
417c417
<         self.Close()
---
>         self.Destroy()
438,439d437
<     debug_dialog("wire_it_callback init complete...")
<
441,443d438
<
<         debug_dialog("wire_it_callback case #1...")
<
450d444
<         debug_dialog("Net namer completed...")
452d445
<             debug_dialog("No net name was entered...")
456d448
<             debug_dialog("An existing net name was entered...")
460d451
<             debug_dialog("A new net name was entered...")
465d455
<         debug_dialog("Adding pads to net...")
469d458
<         net_namer.Destroy()
472,474d460
<
<         debug_dialog("wire_it_callback case #2...")
<
493d478
<         net_namer.Destroy()
496,498d480
<
<         debug_dialog("wire_it_callback case #3...")
<
511,513d492
<
<         debug_dialog("wire_it_callback case #4...")
<
533d511
<         net_namer.Destroy()
536,538d513
<
<     debug_dialog("wire_it_callback building list of nets...")
<
540,542d514
<
<     debug_dialog("wire_it_callback recalc rats nest...")
<
544,546d515
<
<     debug_dialog("wire_it_callback refresh...")
<
549,550d517
<     debug_dialog("wire_it_callback DONE!")
<
```

xesscorp, 2020-05-18:
There actually is a substantive change: the call to Destroy() was replaced with Close() in the callback routine and then Destroy() was called after the name of the net was extracted from the object. I think the destruction of the object prior to accessing the net name was causing a memory error. I'll remove the debugging statements and update the release.

subatomicglue, 2020-05-18:
Awesome!   Sounds like progress. I’ll test when you post an update. Thank
you!

xesscorp, 2020-05-19:
Here is the cleaned-up version. It seems to work OK on my linux system. If it works OK for you, then I'll release it.
[WireIt.py.gz](https://github.com/xesscorp/WireIt/files/4647719/WireIt.py.gz)

subatomicglue, 2020-05-19:
Just annotated all the net names for my pcbnew-only layout.  Seems to work!  
Valuable tool.  Thank you!

xesscorp, 2020-05-19:
I'll push out a new release. Thanks for working with me on this.

closed 2020-05-19 by xesscorp