devbisme/WireIt #14: Cannot start

holmesrichards, 2020-07-23:
**Describe the bug**
WireIt plugin does not start

**To Reproduce**
Steps to reproduce the behavior:
1. Copy WireIt.py and WireIt_icons into either ~/.kicad/scripting/plugins/ or ~/.kicad_plugins/
2. Start Pcbnew
3. Tools > External plugins... > Refresh plugins
4. Tools > External plugins...

**Expected behavior**
Expected WireIt to appear as an available plugin tool

**Computing Environment**
Application: Pcbnew
Version: 5.1.6-c6e7f7d~86~ubuntu18.04.1, release build
Libraries:
    wxWidgets 3.0.4
    libcurl/7.68.0 OpenSSL/1.1.1f zlib/1.2.11 brotli/1.0.7 libidn2/2.2.0 libpsl/0.21.0 (+libidn2/2.2.0) libssh/0.9.3/openssl/zlib nghttp2/1.40.0 librtmp/2.3
Platform: Linux 5.4.0-40-generic x86_64, 64 bit, Little endian, wxGTK
Build Info:
    wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.24
    Boost: 1.65.1
    OpenCASCADE Community Edition: 6.9.1
    Curl: 7.58.0
    Compiler: GCC 7.5.0 with C++ ABI 1011

Build settings:
    USE_WX_GRAPHICS_CONTEXT=OFF
    USE_WX_OVERLAY=ON
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=ON
    KICAD_SCRIPTING_PYTHON3=ON
    KICAD_SCRIPTING_WXPYTHON=ON
    KICAD_SCRIPTING_WXPYTHON_PHOENIX=ON
    KICAD_SCRIPTING_ACTION_MENU=ON
    BUILD_GITHUB_PLUGIN=ON
    KICAD_USE_OCE=ON
    KICAD_USE_OCC=OFF
    KICAD_SPICE=ON

**Additional context**
I had WireIt working before but now it does not. I did upgrade to a new version of Linux Mint recently, don't know if that coincides with when it stopped. I have one other plugin (SchematicPositionsToLayout) in the .kicad/scripting/plugins directory and Pcbnew does find that and it appears in the external plugins list, but not WireIt.

xesscorp, 2020-07-28:
I just tried with the latest WireIt:

1) `cd ~/.kicad_plugins`
2) `git clone git://github.com/xesscorp/WireIt`
3) `kicad &`
4) Open PCBNEW.
5) Tools => External Plugins... => Refresh Plugins... // WireIt (WireIt was already present without the refresh).

So I'm not sure why WireIt isn't appearing for you. I am running Ubuntu 20.04. Here's my system info:

```terminal
Application: KiCad
Version: 5.1.6-c6e7f7d~87~ubuntu20.04.1, release build
Libraries:
    wxWidgets 3.0.4
    libcurl/7.68.0 OpenSSL/1.1.1f zlib/1.2.11 brotli/1.0.7 libidn2/2.2.0 libpsl/0.21.0 (+libidn2/2.2.0) libssh/0.9.3/openssl/zlib nghttp2/1.40.0 librtmp/2.3
Platform: Linux 5.4.0-42-generic x86_64, 64 bit, Little endian, wxGTK
Build Info:
    wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.24
    Boost: 1.71.0
    OpenCASCADE Community Edition: 6.9.1
    Curl: 7.68.0
    Compiler: GCC 9.3.0 with C++ ABI 1013

Build settings:
    USE_WX_GRAPHICS_CONTEXT=OFF
    USE_WX_OVERLAY=ON
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=ON
    KICAD_SCRIPTING_PYTHON3=ON
    KICAD_SCRIPTING_WXPYTHON=ON
    KICAD_SCRIPTING_WXPYTHON_PHOENIX=ON
    KICAD_SCRIPTING_ACTION_MENU=ON
    BUILD_GITHUB_PLUGIN=ON
    KICAD_USE_OCE=ON
    KICAD_USE_OCC=OFF
    KICAD_SPICE=ON
```

closed 2020-08-10 by xesscorp

holmesrichards, 2020-08-13:
I did _exactly_ that and WireIt does not appear.
[image]

xesscorp, 2020-08-14:
I just repeated the install process in my local .kicad_plugins directory and the icons appeared. Note that you have to *execute* the WireIt plugin; the icons don't appear when PCBNEW first starts.

I also found the icons don't appear the *second* time you start PCBNEW. You have to restart KiCad and then the icons will appear on the first invocation of PCBNEW. I have no idea what that's about.

holmesrichards, 2020-08-14:
I can't execute WireIt because it does not appear in the plugins submenu, even if I do refresh plugins. The one other plugin I have installed does appear.

xesscorp, 2020-08-14:
The only thing I can think of is that my version of Python is built by pyenv when it is installed and that rebuilds the wxpython library during the install. I noticed that my Boost and GCC compiler versions are more recent than yours. Maybe that's the difference.

holmesrichards, 2020-08-14:
See above

holmesrichards, 2020-08-14:
I suppose the Boost and GCC versions shown by KiCad are the ones that were used to compile KiCad; they aren't the current ones on the system. I have GCC 9.3.0 installed and Boost I don't think is installed at all. Our PCB revision numbers are the same but mine evidently was compiled for Ubuntu 18 and yours for Ubuntu 20. This Ubuntu 18 version is what I get if I follow the instructions to install KiCad at https://kicad-pcb.org/download/linux-mint/ which say to get it from ppa:kicad/kicad-5.1-releases.